UBC-Snowbots/Snowflake #53: Localisation

garethellis0, 2016-11-26:
**This issue will likely be split off into multiple issues later on, as we decide on how to split up projects**

- full localisation integrating IMU, Lidar, Encoders
- will likely be implemented with some sort of Kalman Filter (either extended or unscented) though research into this still needs to be done
- **TODO** finish writing me

garethellis0, 2017-01-22:
@lkhume This is where I'd like to keep discussion of the new localization system (mostly for documentation and organization purposes). If you've anything you want me to see, just add @garethellis0 to it, and i'll get notified (some goes for anyone else, like val)

ValRat, 2017-10-14:
Obsolete, make a new issue for this.

closed 2017-10-14 by ValRat

garethellis0, 2017-10-21:
*Note:* Closed in favor of #230