UBC-Snowbots/Snowflake #68: Come up with a better dangerScore function for LidarObstacle

garethellis0, 2017-01-23:
- Currently is just `cos(the angle of the obstacle relative to the robot) * (1/the distance of the obstacle from the robot)`
- is in `src/decision/LidarObstacle.cpp`

garethellis0, 2017-02-19:
@raadkhan - Just to reiterate over what we talked about today:

- Figure out a better function for determining the danger score of a given obstacle in `src/decision/LidarObstacle.cpp`
- Write some tests for it in `src/decision/test/lidar-decision-test.cpp`, following the tests that are already there as guides
- Make sure your code compiles and your tests run
- Commit your code and open up a new pull request, then add Val and I as reviewers

Also, great work with the IMU today! :smile:

raadkhan, 2017-02-19:
> Assigned #68 <https://github.com/UBC-Snowbots/IGVC-2017/issues/68> to

Alright 🤔 sounds like a plan.

Again thanks for the help Gareth, you're the man

raadkhan, 2017-02-25:
@garethellis0, what exactly does `sin(the angle of the obstacle relative to the robot) * (1/the distance of the obstacle from the robot)` return? What does it mean mathematically?

garethellis0, 2017-02-25:
@raadkhan - the most dangerous obstacles are the ones right in front of us, and close to use. So I just kinda choose something that gave a lower score to objects farther away (`1/the distance of the obstacle from the robot`) and a lower score to obstacles that are not directly in front of us (`sin(the angle to the obstacle)` will yield a lower score with a greater absolute angle -  keep in mind that directly in front of the robot is 0 degrees)

garethellis0, 2017-03-10:
Just for future reference, if you put something like "closes #35" in your PR message, Github will automatically close the issue when your PR has been merged. (https://help.github.com/articles/closing-issues-via-commit-messages/) (I only discovered this a few weeks ago)

garethellis0, 2017-03-10:
closed via #112

closed 2017-03-10 by garethellis0