UBC-Snowbots/Snowflake #101: Warning Error being thrown when tests are run

garethellis0, 2017-02-19:
There is an `ROS_WARN` being thrown when the tests are run, namely:
```
[ERROR] [1487482866.179100462]: Given odd length list of waypoints, check that "waypoints" parameter has an (numerically) even length. As this is a series of lat/lon coordinates,it should always be even
```
This is an expected error, originating from the fact that the test causing it is actually testing an odd number of points. This error is causing warnings in the tests, and so should somehow be removed, or modified so that it indicates that it is acceptable for this to fail in testing (ie. add `[t is acceptable for this to fail in testing]` to the end of it, or something like that)

raadkhan, 2017-03-28:
Just to clarify, am I supposed remove the test failures simply by modifying ```EXPECT_EQ(0, command.linear.x);``` such that it evaluates as true and then comment ```it is acceptable for this to fail in testing```?

raadkhan, 2017-03-28:
@garethellis0 I have been attempting to actually fix the test issues by first finding where the error is coming from. I did not notice anything odd except in ```oneObstacleStraightAheadTest``` ```obstacleToLeft``` and ```obstacleToLeft``` all call ```std::fill( )``` and are these are the three tests which fail, but ```noObstacles``` does not call ```std::fill( )``` and is the only test that passes. Specifically, I see that changing the value to be populated in the range (third parameter) changes ```command.linear.x``` and command.linear.z but I cannot a see a relation... so how to make the assigned value such that ```command.linear.x``` evaluates to = zero and ```command.linear.z``` evaluates to >= 1e-5

garethellis0, 2017-03-29:
@raadkhan - this issue has doesn't have anything to do with the problem you're experiencing, so I'm going to move discussion over the the PR.