UBC-Snowbots/Snowflake #110: Calibrate the Razor IMU

garethellis0, 2017-02-27:
We've just received our brand spanking new IMU! It's pretty cool, lots of pretty accurate data, and all in a neat little package. However, it still needs to be calibrated. Calibration instructions can be found (here)[http://wiki.ros.org/razor_imu_9dof] (under the `indigo` section). The determined values should be written to our copy of the firmware file in `src/firmware/razor_imu_9dof`, as per the calibration instructions.

raadkhan, 2017-02-27:
> Assigned #110 <https://github.com/UBC-Snowbots/IGVC-2017/issues/110> to
> @raadkhan <https://github.com/raadkhan>.

Got it 👍

garethellis0, 2017-02-27:
@raadkhan - just a quick note that this is on hold until I merge the gps_driver code (currently no PR open yet), as I moved the location of the firmware code on the same branch

garethellis0, 2017-03-05:
@raadkhan - just put it on it's own branch and open a PR for it, I can merge it to master and then merge master back into  other branches if I need it

garethellis0, 2017-03-05:
Closed via #118

closed 2017-03-05 by garethellis0