UBC-Snowbots/Snowflake #123: Create  a "New member setup readme"

garethellis0, 2017-03-09:
Should be at root, separate readme from `README. md`. Should walk new members through a few tutorials to help them get up and running. First tutorial should be git. @raadkhan, what was the git tutorial you used?

raadkhan, 2017-03-09:
@garethellis https://try.github.io/levels/1/challenges/1
This one is also kind of nice using GUI:
https://guides.github.com/activities/hello-world/

garethellis0, 2017-09-09:
@RobynCastro @raadkhan - could you guys post some suggestions here for stuff that you found useful? 

@raadkhan - I'm assigning this to you because you went through the onboarding process the most recently, and so are most likely to understand any issues that new members may face. Try and get this done before orientation sept 23 if possible. Let me know if you have questions

garethellis0, 2017-09-09:
Some suggestions (Feel free to just edit this comment to add more):
- Rarely will github issues contain all the information that you need to resolve them. Don't be afraid to ask for help, clarification, or just more information.

raadkhan, 2017-09-11:
@garethellis0 is it possible to get access to the software challenge page on the snowbots website (looks like the website is being updated and that page is not there now)?

garethellis0, 2017-09-11:
Just to confirm, the top taskbar is broken on mobile on the website? If so, could you open an issue on our website repo https://github.com/UBC-Snowbots/UBC-Snowbots.github.io/issues  to see if @SimonJinaphant can fix this?   Thanks for noticing that, good catch!

raadkhan, 2017-09-12:
done by @SimonJinaphant

PR #235 (2017-10-09): Created readme for new members
resolves #123
closed 2017-10-07 by raadkhan