UBC-Snowbots/Snowflake #127: Zed submodule not being blacklisted in CLion

garethellis0, 2017-03-12:
- Currently, I believe that CLion is using the old `catkin_make` command or equivalent to build. 
- We have migrated to using `catkin_tools`, with the `catkin build` command, as it allows us to blacklist packages that we might not be able to build from being built along with everything else
- **We need to**: Figure out how to make CLion use the `catkin build` command

**Current Workaround**: create an empty file called `CATKIN_IGNORE` in the `src/zed-ros-wrapper`,  or just don't clone with submodules.

garethellis0, 2017-03-31:
@ValRat - can you confirm that switching over to the old build tools has fixed this?

garethellis0, 2017-09-09:
@ValRat - please see above comment

garethellis0, 2017-09-27:
@ValRat - :point_up: :point_up: :point_up:

ValRat, 2017-09-27:
Will look at tonight

garethellis0, 2017-09-28:
This isn't ideal, but I'm OK with it for now, mostly because I don't see any way to a better solution.

closed 2017-09-28 by garethellis0