UBC-Snowbots/Snowflake #141: Look at switching to industrial_ci for CI

garethellis0, 2017-03-31:
[industrial ci](https://github.com/ros-industrial/industrial_ci/) seems like a pretty solid CI tool for ROS, and supposedly covers kinetic. Test, and if it seems to work well, consider using.

garethellis0, 2017-03-31:
Note that industrial ci is far more complex then  our current system, and so may be complete overkill, AND make it harder to manage.

130s, 2017-12-13:
I'm one of the authors of industrial_ci and just randomly hit this ticket.
It may seem complexed indeed but you can start with very simple setting. I bet majority of the users use only the minimal feature. Take a look at [Quick Start setcion](https://github.com/ros-industrial/industrial_ci#quick-start). And, we'd like to hear whatever issues you have. It's actively maintained. Thanks.

garethellis0, 2017-12-14:
@130s thanks for taking the time to comment here. :raised_hands:  I'll have some free time soon, and so will take a look at industrial_ci again. As a note the only feature that I might see blocking us is the lack of clang-format support. Which I see is on your (roadmap)[https://github.com/ros-industrial/industrial_ci/issues/98] but does not yet seem to be implemented.

garethellis0, 2018-01-04:
Just taking another quick look at this, it is worth noting that our current CI system has actually started to become quite a bit more complex then was originally intended, and has diverged from the base [moveit_ci repo](https://github.com/ros-planning/moveit_ci), so this change over doesn't seem that unreasonable.