UBC-Snowbots/Snowflake #142: Test (and hopefully fix) the compass on the razor imu

garethellis0, 2017-03-31:
- Currently, the orientation data we're getting from the razor IMU is basically useless. It doesn't seem to be pointing in the right direction, and will not change even **remotely** the right amount when rotated.
- I suspect that this is due to the compass being either broken, or completely miscalibrated

**TODO:**
- Test the razor IMU in several different environments to ensure that the bad behavior is not just being caused by things in the local environment, just as large chunks of metal
- Attempt to fix it. A good starting point for this would be to properly calibrate it [here](https://github.com/KristofRobot/razor_imu_9dof#calibrate). If we are unable to fix this, inform Gareth or Val, so we can get a replacement **ASAP**

**NOTE:** This is time sensitive. This is a extremely important component, and if it's not working, and if we can't find a substitute, nothing will run.

raadkhan, 2017-04-01:
- accelerometer and gyroscope calibrated
- magnetometer on hold

garethellis0, 2017-09-09:
We have a new IMU, and will not be using the razor this year, so this is no longer relevant.

closed 2017-09-09 by garethellis0