UBC-Snowbots/Snowflake #158: Grab testing data and store it for future testing

ValRat, 2017-04-14:
Consider creating a folder with input data from sensors: camera, gps, etc. for testing purposes. Preferably this will be in rosbag formats for easy consumption - host this in a share somewhere and not the repo to avoid people downloading 1GB of bagfiles needlessly.

ValRat, 2017-10-14:
On another note, ROSbags do get pretty big (1GB/10 seconds of zed data).
Could consider buying an external 1TB to store it in and people pull in data as they need.

garethellis0, 2017-10-21:
I'm a bit against storing things on a physical external drive, as it's much more likely to get lost or disorganized. Perhaps we could reach out to google for some sponsorship help there?

ValRat, 2017-10-21:
Even still, files can get up to multi gigabytes, would be painful to
download from home.

garethellis0, 2017-10-21:
That's also a good point. Let's discuss this at meeting tomorrow.

garethellis0, 2018-02-11:
@ValRat - we should revisit this issue with the upcoming test cart that the mechanical team is building. I think if we're not capturing zed. For reference, ~30 seconds of rosbag capturing almost all the sensor data from the robot (sans the camera/pointcloud data) is about 3mb.