UBC-Snowbots/Snowflake #171: Create a PointCloud -> Image Node

garethellis0, 2017-05-01:
- We would like to be able to take the filtered and flattened pointcloud that we get from `zed_filter` and convert it to an image that vision can use.
- Since the pointcloud will already be transformed to the robots frame of reference, getting a birds eye view should be as simple as taking each point and making it a pixel.e
- This image can then be used by `vision_decision`

garethellis0, 2017-09-27:
@ValRat - do you think we need this, given the plans to do a more advanced system this year (with basic mapping, etc.)?

ValRat, 2017-09-28:
We will still do filtering on the output, just not the one described here. I think it's safe to close this issue and recreate it when we know exactly how we want to filter/consume the zed data with respect to the advanced system.

closed 2017-09-28 by ValRat