UBC-Snowbots/Snowflake #176: Add Arduino To Install Scripts

garethellis0, 2017-05-06:
- We need to add the `flushRX` commands to `HardwareSerial.h` before we compile some firmware (firmware_control **cough cough**)
- The exact changes are [here](https://forum.sparkfun.com/viewtopic.php?f=32&t=32715%20Serial.flushRX()) but it's basically adding this to `/usr/share/arduino/hardware/arduino/cores/robot/HardwareSerial.cpp`:
```
void HardwareSerial::flushRX()
{
  _rx_buffer->head = _rx_buffer->tail;
}
```
and adding this to `/usr/share/arduino/hardware/arduino/cores/robot/HardwareSerial.h`
```
virtual void flushRX(void);
```

- We need to add the arduino installer to the install script and then apply a git patch file to these files to make the required changes (or maybe just append them :laughing: )

Dulluhan, 2018-02-10:
If this is done, we should close this

garethellis0, 2018-02-11:
Good catch, thanks!

PR #265 (2018-01-20): Added arduino to the setup script
resolves #176