UBC-Snowbots/Snowflake #180: sb_util handling templated functions

ValRat, 2017-05-13:
`sb_utils` contains the function `SB_doTransform` which then calls another template function `tf2::doTransform()`. The issue is the calling class of `SB_doTransform` must link to the proper implementation of `tf2::doTransform()` for the proper datatype (i.e. `sensor_msgs::Pointcloud2`).

We should probably link all of the common `tf2::doTransform()` implementations to `sb_utils` to reduce overhead of the calling classes making it easy and everything should just work (tm).

Note:
Custom datatypes can also leverage `tf2::doTransform()` by implementing it. See [link](http://wiki.ros.org/tf2/Tutorials/Transforming%20your%20own%20datatypes).

garethellis0, 2017-12-02:
`SB_doTransform` was removed, as it was *not* the right way to use the ROS TF tree. Closing.

garethellis0, 2017-12-02:
Whoops, I thought it had been removed, but looks like not. @ValRat, what are your thoughts on keeping it around?

PR #256 (2017-12-02): Sb utils transform deprecation
Close #180
closed 2017-12-02 by ValRat