UBC-Snowbots/Snowflake #199: Consider Merging IARRC-2017 and IGVC-2017 into the same repo

garethellis0, 2017-09-10:
Much of the code, especially for vision processing, is shared, and it would make sense to have them in the same repo, with competition specific nodes labelled as such.

garethellis0, 2017-09-10:
Note this would also mean we should name the repo to something like `Software`, thus also closing #200

garethellis0, 2017-09-13:
- our CI workflow is repo agnostic (it should work for any ROS repository with a setup script of the same name). So CI should be fine (it worked totally fine for the IARRC repo, despite being created way before it). Likewise, building should be fine.
- it's definitely a more complicated git workflow, which isn't great :cry: but not one that everyone needs to understand, just those dealing with merging code between repos.

- I actually think we should backup the `Snowflake` repo (rename it to something like `Snowflake Backup`, and then use the name `Snowflake`, mostly because I think it's rather appropriate (ie. this is our "Snowflake" :laughing: ). 
- I think the process here would be:

1) rename `Snowflake` -> `Snowflake_old`
2) rename `IGVC-2017` -> `Snowflake`
3) label igvc specific things (mostly at the package level) in the `Snowflake` repo (ex. `igvc_decision`) (using a prefix instead of a suffix allows for things to be first sorted by competition, then by package)
3) open pull requests on `Snowflake` with useful parts of code from IARRC, labelling them appropriately (ex. `iarrc_drag_race`)

garethellis0, 2017-09-27:
We ended up using the following structure
- One repo (`Snowflake`) with the following branches: `core`, `igvc`, `iarrc`
- `core`: contains any code not specific to one compeition (for example, the vision HSV filter)
- `igvc` and `iarrc` contain all the code in `core` (which has to be kept manually in sync) and any code specific to that respective competition
- all members must have forks of `Snowflake`, and open pull requests from their forks.

closed 2017-09-27 by garethellis0

ValRat, 2017-09-28:
The way of the open source :eyes: