UBC-Snowbots/Snowflake #225: Create Node to filter Pointcloud based on height from ground level

garethellis0, 2017-10-06:
We would like to be able to filter points the in the pointcloud coming from the Zed based on their height from the ground. The general structure would be:
- take in `sensor_msgs/Pointcloud`
- using the [transform tree](http://wiki.ros.org/tf), translate the pointcloud from it's frame of reference (relative to the Zed) to the frame of reference of the robot
- Filter the points based on their height from ground level (+/- some given deviation)
- publish the filtered cloud

garethellis0, 2018-01-13:
Interesting note that @ValRat made over on #226:
"After rolling out my own filter node I found out that: 
`pcl_ros` provides nodelets for various [filters](http://wiki.ros.org/pcl_ros/Tutorials/filters
)."

garethellis0, 2018-03-19:
@ValRat - is this resolved now?

ValRat, 2018-03-19:
Yep

closed 2018-03-19 by ValRat