UBC-Snowbots/Snowflake #227: Create a Node to Extract Lines From PointClouds

garethellis0, 2017-10-06:
We'd like a node that can extract mathematical lines from pointclouds. The general structure here would be:
- take in the pointcloud
- group points that are within a given range of each other
- do a polynomial line of best fit through the points (we should be able to easily change the degree of the polynomial)
- broadcast an `igvc_obstacle_list` message (note that this message type does not yet exist)

garethellis0, 2017-11-25:
An interesting paper on extracting objects from 3D point clouds can be found here: https://arxiv.org/abs/1711.06396

mingyokim, 2017-11-25:
# My approach to this problem so far:
## Assumptions:

* The data points [(x,y,z),(h,s,v)] that do not pass the HSV filter are simply removed from PCL
* Bounding box is described in 2D

## Implementation:

* PCL is in 3D, but what we really want is a 2D projection from above. This can be achieved by ignoring the z component.  
* Remove points from PCL that are outside the bounding box.
* Perform density-based clustering on the resulting PCL. 
  * Rough idea for algorithm: [link](https://www.cs.ubc.ca/~schmidtm/Courses/340-F17/L9.pdf) (taken from my ML class)
  * This would theoretically work well if HSV filtering produces distinct groups of points for each line, but it would be quite sensitive when the boundaries between the groups are ambiguous.
  * We would have to fine-tune the hyper parameters for minimum neighbors and radius by observing its performance on sample data. 
  * We may also have an expected number of clusters (most likely 1 or 2 lines?); in the case where we have more clusters than we expected, we can just keep the clusters that contain the highest number of points and discard the rest. In the case where we have less clusters than we expected, we could re-run density-based clustering  with stricter parameters (for minimum neighbors and radius), but this is probably computationally inefficient. 
* Perform non-linear regression by means of change of basis (this can take in p for degree of polynomial so we can easily adjust this) for each cluster. However, keep in mind that the number of basis grows exponentially with the degree of polynomial (O(2^p)). If we want to use a higher degree of polynomial, we would need to use kernels but I'm assuming we would probably stay around p=3.

garethellis0, 2017-11-30:
This seems like a really well thought-out approach to the problem. I like the idea of filtering out clusters, another possible idea for doing do so might be to have some sort of density threshold for the them, which would allow for any number of lines (we might only be able to see one, or we might be able to see more then two if one leaves and re-enters the pointcloud).

### Generating Test Data
- since #226 and #225 aren't done yet, it's important that we have some way of generating psuedo-realistic testing data
- one option might be to create a node to generate a pointcloud from a given set of polynomial lines by generating a random distribution of points around them 

Excellent work here! I'm excited to see how this progresses :clap:

mingyokim, 2018-01-06:
## Approach to Generating Test Data
Given a bounding box with (x1, y1) and (x2, y2), and a polynomial line:
* Starting from x1, increment x by deltaX until x2
* Calculate y from x, and disregard when y is outside of the box
* Generate N number of random points in a small, square box of size MxM around the point (x,y)
  * Could also do it by radius instead of an MxM box, but box seems more straightforward

commit a09a17c2 (2018-03-16): Extract lines from PointCloud (#278)

Resolves #227.
Created a new package named 'sb_pointcloud_processing', with a node called 'LineExtractorNode'.
The LineExtractorNode makes use of DBSCAN and Regression classes which were also added for this commit.
Note that the input for LineExtractorNode is a filtered PointCloud by #284, which is not implemented yet.
closed 2018-03-16 by mingyokim