UBC-Snowbots/Snowflake #233: Create a Node to Convert a Path to a Twist

garethellis0, 2017-10-06:
We would like a node to convert a given path into a Twist message to send to the robot. The general structure would be:
- take in a path
- compute a twist message from the path
- broadcast the calculated twist message

[image]

garethellis0, 2018-03-19:
@WilliamGu98 - is this resolved now?

WilliamGu98, 2018-03-19:
Yeah I just merged it yesterday

garethellis0, 2018-03-19:
Great! Just a note for the future, you can automatically close issues with a pull request, see here for details: https://help.github.com/articles/closing-issues-using-keywords/

closed 2018-03-19 by garethellis0

WilliamGu98, 2018-03-19:
Alright thanks, I'll take note of that