UBC-Snowbots/Snowflake #250: Improve the Twist Message -> Actual Movement Mapping in Olaf Firmware

garethellis0, 2017-11-04:
Currently, we just have an arbitrary relationship between the `angular_z` and `linear_x` and the actual turn rate and motor speed. `angular_z` is supposed to be in rad/s and `linear_x` is in m/s. 

There are two main points to be done here:
- implement some sort of relation between `angular_z` and `angle` under the `drive` function, assuming `angular_z` is in rad/s, such that we turn at the expected rate (up to maximum the servo turning the wheels can go)
- implement some sort of relation between `linear_x` and `velocity` under the `drive` function, assuming `linear_x` is in m/s, such that we move at the expected rate

marinah1, 2018-08-01:
@garethellis0 Pushed firmware onto https://github.com/marinah1/Snowflake

raadkhan, 2020-02-08:
Can reopen if we decide to ressurect Olaf

closed 2020-02-08 by raadkhan