UBC-Snowbots/Snowflake #304: .rosinstall doesn't seem to 'install' packages as we think it does

ValRat, 2018-04-21:
Looks like `rosinstall` is more of a scv agnostic source file grabber. We can keep using it, but proposing that we put the repositories pulled under `/src/external_pkgs` so that all gets built locally using `catkin_make`.

garethellis0, 2018-04-24:
I really like this idea! @Dulluhan - you've got the most experience with this stuff currently, what do you think?

Dulluhan, 2018-04-24:
Unfortunately it seems like the only solution right now. This also means we will have to look ahead and figure out how we want to install the dependencies for certain packages. Maybe changing the order of how we run the `install_depdendencies` script and run `rosinstall` before `rosdep update`?

ValRat, 2018-04-27:
To clean up the discussion, `wstool` is the next iteration of `rosinstall` and the changes in the code have been made to reflect that.

closed 2018-04-27 by ValRat

Dulluhan, 2018-04-29:
I was under the impression that `rosinstall` replaced `wstool` :open_mouth:

PR #303 (2018-04-27): Add wrapper submodule
Rosinstall used as a source file grabber, fixes #304 
added realsense camera module, fixes #302