UBC-Snowbots/Snowflake #343: Create a ROS Abstraction for Radio Communications

RobynCastro, 2018-09-08:
We need to be able to connect to the robot like it's another computer.

garethellis0, 2018-09-08:
In general, if you have two computers running on the same network, it's fairly trivial to run ROS nodes on both computers (see: http://wiki.ros.org/ROS/Tutorials/MultipleMachines). The real challenge here is being able to have multiple computers on the same network over a radio link. We could potentially use something like: https://www.avalan.com/products/900mhz/aw900xtr-outdoor-wireless-ethernet-radio/, but it might be expensive, and we have to keep our price cap in mind (even sponsored items count towards the price cap).

garethellis0, 2019-02-03:
This looks good to me! I think [rosserial](http://wiki.ros.org/rosserial_embeddedlinux) is a great option, sounds like people have got it to run on the `WRT54GL` router.

raadkhan, 2019-02-04:
Alright nice! I'll look into `WRT54GL` and propose it to the rest of the team

raadkhan, 2019-10-05:
Resolved with NM-2450's wide operating range and simply communicating with multiple machines by design in ROS

closed 2019-10-05 by raadkhan