UBC-Snowbots/Snowflake #387: Complete the networking system

ihsan314, 2020-09-17:
Kevin, Raad and I have been fiddling with the team's routers, trying to find optimal settings that maximize range. Additionally Raad and I have two branches [here](https://github.com/raadkhan/Snowflake/tree/feat-rf-comm-server) and [here](https://github.com/ihsan314/Snowflake/tree/ihsan314-rf-comm) that contain ROS nodes that can communicate over a network.

To extend this, what needs to be done is making sure the connection between the routers is robust (we may need to send video feeds and will certainly be sending ROS commands over the link), and abstracting out any settings we do make for the routers so that our code doesn't break easily if we need to change our router setup.

ihsan314, 2020-10-20:
An [old pull request](https://github.com/UBC-Snowbots/Snowflake/pull/378) does something similar to what my branch and Raad's branch do. Both branches are linked above.

In terms of verifying the current router setup, we will need to spec another battery pack. Currently we have the following two battery packs: [one from Intocircuit](https://www.amazon.com/Protector-Tempered-AVIGOR-Coverage-Friendly/dp/B00BB5VQCE) and [one from MAXOAK](https://www.amazon.com/MAXOAK-50000mAh-Portable-Charger-External/dp/B00YP823NA/). The latter one is faulty, as it doesn't retain charge well anymore. so it needs a replacement. It's unclear what decision process led to the two battery packs here. Uncovering that will be nice as it will aid in selecting a viable replacement for the MAXOAK battery pack. If not, then a new decision process will need to be made based on the power consumption ratings of the routers selected, and their ability to interface with the barrel tips on the routers.

ihsan314, 2020-12-29:
Today I revisited the routers and determined what settings I enabled previously.

The TP-Link router has an IPv4 address of 172.16.16.155. The Doodle Labs router has an IPv4 address of 172.16.16.1. Both have a netmask of 255.255.255.0, and do not have DHCP enabled. Thus one must manually assign static IP addresses to connected endpoints.

For endpoints connecting to either router, IP addresses of the form 172.16.16.x are valid as long as x is not 0, 255, 1, or 155, and as long as another endpoint is not using the chosen value of x.

raadkhan, 2021-02-13:
Research shall be documented here for now: https://docs.google.com/document/d/1mezcO8ChN-ejV8PdgcQNHCxefTBNUvT-mbBezU3Kgv4/edit

ihsan314, 2021-07-18:
To successfully run ROS nodes on two separate computers on the same network, one computer should be chosen to run `roscore`. Both computers should export their IP address to the variable `ROS_IP`. The computer not running `roscore`   should export `ROS_MASTER_URI=http://<ip-address-of-other-machine>:11311`.

ihsan314, 2021-08-08:
To set a static IP address in Ubuntu, one should save the following lines into a file named `/etc/netplan/99_config.yaml`:
```yaml
network:
      version: 2
      renderer: networkd
      ethernets:
              wlp59s0:
                      dhcp4: no
                      addresses:
                              - 172.16.16.3/24
                      gateway4: 172.16.16.155
```
The addresses `172.16.16.3` and `172.16.16.155` and the interface `wlp59s0` should be replaced with appropriate values. interface `

closed 2022-09-05 by ihsan314