UBC-Snowbots/Snowflake #412: Detect when we have observed a QR code or a particular marker type

ihsan314, 2022-07-15:
We have two Intel Realsense cameras publishing to the topics `/cam1/color/image_raw/compressed` and `/cam2/color/image_raw/compressed`.

During competition, these cameras may potentially be pointed towards markers from a dictionary of 50 4x4 ArUco markers or QR codes with a status message.

We need to create a node or two that will subscribe to the camera feeds, scan for a marker/QR code, and report back with the marker numbers or the contents of the QR codes found.

OpenCV is known to [cooperate with ROS](https://wiki.ros.org/cv_bridge/Tutorials) and can handle the [detection of ArUco markers](https://docs.opencv.org/4.x/d5/dae/tutorial_aruco_detection.html) as well as the [detection of QR codes](https://docs.opencv.org/4.x/de/dc3/classcv_1_1QRCodeDetector.html). So this task can likely be accomplished with OpenCV.

Roozki, 2022-07-19:
looking into scanning codes with OpenCV, it has a built in package that scans aruco codes so this issue shouldn't be too difficult. However we should consider ways of making sure the cameras are pointing at the codes.
I'm thinking:
1. proximity to the GPS waypoint will trigger the scanning node
2. then the node checks if there is a code and attempts to scan
3. if no codes are found, the driveterrain rotates the camera
4. repeat 2-3, but make sure to "time out" if no codes are found

ihsan314, 2022-07-19:
That's a lot smarter than continuously scanning, which was what I was initially thinking.

1 can be accomplished by publishing to a topic.

2 can be done by looking at the most recently received frames from either camera and checking if a marker or code has been found

3 can be done by spinning the rover slowly.

closed 2022-09-05 by ihsan314